TochigiRaspberryPICircle/kit #1: Smalrubyの準備

vestige, 2017-01-28:
* Mac, ラズパイともインストール失敗
* 会社のwindowsはインストール、実行まで完了
* この後の選択肢
   * 誰かのwindowsにインストールするか？
   * Rubyのコードが出力できるところまでするか...
   * 別のにするか

mame-n, 2017-02-03:
この前、toRubyで見た時は、限定かもしれないけど動いていたよね。

vestige, 2017-02-03:
そですねー。できれば実際にうごいてるとこまでみせたいですねー

naominix, 2017-02-04:
Smalrubyですが、

Raspberry Pi 3(NOOBS2.1.0のRaspbian）
Ruby2.2.2(rbenv)

で動作しました。

Ruby2.4.0ではNGでした。

vestige, 2017-02-05:
おー、ありがとうございます！
このあと、無事にねこさん、うごきました？

naominix, 2017-02-05:
はい。ネコさんも他のキャラも動きました。

vestige, 2017-02-05:
おー、よいですねー。
なんで自分のだめなんだろう...
もう一度チャレンジしてみます。

vestige, 2017-02-05:
@naominix さんはこの手順でやったのでしょうか？

https://docs.google.com/document/d/1USIWn9YUdRqiquAgOXx_tkWhrZA48brqxZ-HmVFX_Kg/mobilebasic

naominix, 2017-02-05:
この文書は初めて見ましたが、やった手順的には近い気がします。

vestige, 2017-02-05:
僕のはmacでやっても、ラズパイでやっても同じようなエラーになってしまうんですよね。
dxruby_sdlが正常にインストールされてないような感じがするんですよね...

naominix, 2017-02-05:
dxruby_sdlは割りとすんなり入った記憶が(^-^;;

vestige, 2017-02-05:
インストールはすんなりなのですが、実際に動かすと動かないので、困っている感じなのですー。

naominix, 2017-02-05:
なるほど。どんなエラーが出るのでしょうか。

vestige, 2017-02-05:
あ、いま入れ直して見てるので、またエラーでたら貼ってみますー。

vestige, 2017-02-05:
https://github.com/TochigiRaspberryPICircle/kit/blob/master/screenshot/0205.jpg

naominix, 2017-02-05:
なんと、起動後に出るんですね（驚）。
これは出ないでちゃんとネコも他のキャラも動いていますね・・・。
何が違うんだろう・・・

vestige, 2017-02-05:
そなんですよ。なにかSDLまわりでインストールをしくじってるんですかね。よくわからないんですー。

naominix, 2017-02-05:
もう一度、NOOBS2.1.0を焼き直して最初からやり直してみました。

教えていただいたドキュメント
https://docs.google.com/document/d/1USIWn9YUdRqiquAgOXx_tkWhrZA48brqxZ-HmVFX_Kg/mobilebasic?usp=gmail

の以下の部分だけ飛ばしてセットアップを実施しました（以下の部分は以前もやった記憶が無かったためです）

----飛ばした手順----
LXTerminalを起動して、
$ update_rubygems
$ gem update
→(H27.9現在で、2.0.0-p353ではrakeとrdocの上書きは「N」を選択。2.2.2ではプロンプトが出ない)
$ rbenv rehash
☆dxruby_sdlのインストール
----飛ばした手順----

その結果、問題なく動作しました。
Audio周りの設定は特にいじっていないのでデフォルトのままですね（HDMI出力になっている状態です）。
ブロックを移動してはめる時のカチって音がなるのでここが動くかどうかが問題になっているのかもしれませんね。

vestige, 2017-02-12:
まっさらからインストールしなおして動くようになりました。
おそらくpythonでゲームライブラリを使った時にインストールしたライブラリのどれかが問題だったみたい。
依存がおおくて複雑すぎる...

closed 2017-02-12 by vestige