TochigiRaspberryPICircle/kit #4: 2月9日夜のスタッフミーティング

mame-n, 2017-02-03:
2月9日のスタッフミーティングで確認したいことありませんか?
というか、その場で決めていく感じかも。

- 当日の公共交通手段はあるのか。(つまり送迎バス)
- 参加表明について
- 当日の開催時間、当日の準備時間、撤収時間
- 使用可能な機材、準備しなければならない物

他にあれば連絡チョーダイ。

vestige, 2017-02-03:
とりあえず、そんな感じかもですね。
中内さんが描いた当日のイメージが星野さんのイメージとあってるかを確認ですかね。
借りれる機材はあるのですかね？

naominix, 2017-02-03:
まだ具体的なことは決まっていない状態なので、決められることをどんどん決めていきたいと思います。

規模としては小さめなScratchDayを想定しています

元々ボランタリーなイベントなので、一般の方向け送迎バスについては難しいかもしれませんが、那須町内の参加希望の子ども達を主対象にスクールバスの運行が出来ないかを打診していますが実現できるかは分かりません。

機材については若干古いノートPCを複数台借りることができる見込みです（九尾祭りで使用したものなど）。

那須ラボ保有のわずかなPCも持っていく予定です

mame-n, 2017-02-03:
プログラミングデーin那須町のように、事前に参加希望を募っちゃうのもいいかもしれないですね。
人数とどこから来るかがわかれば、スタッフと乗り合いで来てもいいですしね。

機材はSmalrubyがOSXやRaspbianでうまく動かないようなので、WindowsPCがあればお借りしたいです。

naominix, 2017-02-03:
そうですね。プログラミングデイ同様、事前に参加希望を募る方向でお話はしています。

ちなみにSmalrubyは私の手元のRaspberry Pi 3(NOOBS2.1.0のRaspbian）で動きました。
rbenvでRuby2.2.2で動作しましたのでお知らせしておきます。

mame-n, 2017-02-26:
まだ公の情報ではありませんので、公開はしないでください。

https://docs.google.com/document/d/1_HsiI3W38y9-kK3eIZLNfx9zFTiegZgl3FSaG8uiXr8/edit

akimach, 2017-02-27:
@mame-n アクセス権を要求しました 🙏

vestige, 2017-02-27:
@akimach 勉強会の時に見るのに貼ってもらったやつなんだけど、とりあえずもう少しまっててー

akimach, 2017-02-27:
@vestige 了解っす 👍

mame-n, 2017-03-01:
まだかなり柔らかいので、誤解を与える可能性もあり、toruby内部への公開も、もう少し先にしますね。

akimach, 2017-03-01:
@mame-n 了解です

naominix, 2017-03-06:
すでにご存知かと思いますがこちらにも貼っておきます。

ScratchDay公式サイトに情報を登録して公開しました

ScratchDay in Nasu
https://day.scratch.mit.edu/events/3180/

naominix, 2017-03-06:
リースアップPCについてはESPRIMO D530/A(CPU:Core2 Duo)が１５台程度は確保できるのではないかと思っています（動作確認は１台のみしかしていません）。
小中学生にUbuntu Linuxインストール大会に挑戦してもらって事前に準備できればと考えています。

vestige, 2017-03-07:
こちらのスタッフとしての参加表明をとりまとめておきますねー

vestige, 2017-03-28:
こっちもいったんクローズしておきますね。

closed 2017-03-28 by vestige