TochigiRaspberryPICircle/kit #8: Scratch Day 2017 in Nasu：当日までに用意するもの

vestige, 2017-03-31:
いまのところこんなところ？他にあるかな？

- suganumaさんの発表資料

- takataさんの展示資料

- よねざわのtorpc 紹介資料

- mameさんのゲーム？ #5

mame-n, 2017-04-01:
smalrubyは？

naominix, 2017-04-01:
smalrubyは那須町教委のノートPC（九尾祭りで使用したもの）に導入した方がよろしいでしょうか。

vestige, 2017-04-01:
忘れてました。ラズパイにいれてきますー

naominix, 2017-04-01:
了解です。
ラズパイの周辺機材（ディスプレイ、キーボード、マウス、その他ケーブル類）も大丈夫でしょうか？

vestige, 2017-04-02:
1セットは用意できますが、予備で1セットあると助かりますー

mame-n, 2017-04-08:
星野さんにお願いしたいものを一覧しませんか。
新しいissue立てる？

vestige, 2017-04-08:
そかもですね、一つ立てておきますか

naominix, 2017-04-08:
よろしくお願いします。
ちょっと当初準備できると思っていたノートPC系があやしくなってきているので、再度確認をしておきたいと思います。

mame-n, 2017-05-11:
危険なのは人のをあてにしてるやつです。
私はミルククラウンでHTMIモニターを期待してます。あとは自力でなんとかなる。

naominix, 2017-05-11:
那須ラボの27インチモニターと21インチモニターは持っていくようにしたいと思います。27インチはKinectで（デモは古いモノしかありませんが、そもそも体験したことが無い子も多いと思いますので）。
他に10インチくらいの小さめなHDMIディスプレイなんかもありますが、使えるでしょうか？

naominix, 2017-05-11:
ちなみに那須町教委のノートPCは全く用意出来なくなってしまったのですが、NTTドコモさんの協力でWindowsタブレットを６台お借りしました（CPUが古いATOMなのでかなり非力ですが）。
Viscuitアプリを起動しただけでCPU100％張り付くという軟弱モノですがScratchなら余裕です。
メリットはLTE通信SIMが内蔵されているのでネットが使えます。
もちろん、Hour of Codeなんかも大丈夫です。

naominix, 2017-05-11:
展示の表示の紙なのですが、準備しようと思っていたら全然時間が取れなくて明日は町教委に行かないので指導主事に依頼して印刷していただこうかと思っています（ただし、現行はこれから作ります（汗））。

naominix, 2017-05-11:
電源タップは那須ラボにあるものは出来るだけ持って行きます。
ラズパイ系であれば、ラボで使っているLapdock Atrix 4Gも使えるのであった方がよろしいでしょうか。

fleg949, 2017-05-11:
おはようございます。高井です。
明日は、宜しくお願いします。

明日、自分は、下記のものを持って行く予定です。

mBot ピンク
17インチと23インチモニター
電源タップ  3～4個
ラズパイ2

あまり関係ないかもですが、WioNode、PINE64、C.H.I.Pなども持って行くつもりです。

あと、内履用のスリッパか、靴は、必要でしょうか？

mame-n, 2017-05-12:
ScratchワークショップはWindowsタブレットでやると言うことですね。
ViscuitはiPadですか? ビスケットワールドでしたっけ、広場でしたっけ、その表示にはプロジェクターなのかな?
Chibi:bitのコード書き用PCはどうなりどうでしょう。

テーブルタップはラズパイ勉強会組は余ってもいいから持ってこられるひとは持ってきましょうか。

mame-n, 2017-05-12:
ロボット系を動かす時にiPadあると楽しみが増えると思いますが、用意できそうですか?
OSMO出そうとするとやっぱりiPad必要ですね。

naominix, 2017-05-12:
> 内履用のスリッパか、靴は、必要でしょうか？

こちら、不要です。

mame-n, 2017-05-12:
給湯室で何か飲めるように紙コップとペット数本、買っていきます。

naominix, 2017-05-12:
Scratchワークショップは先程締め切りました。参加者はWebから３名、電話で１名ありましたので４名でした。内容も考え直さないとアカン・・・orz

naominix, 2017-05-12:
Osmoは小物が多いということもあって、見ていられる人がいない関係上、展示を諦めようかと思っています（展示したいのは山々なのですが）

naominix, 2017-05-12:
iPadは
・那須町教委のiPad23台（LTE）
・那須ラボのiPadとiPad mini2で２台（WiFi）
を用意します。
Kindle Fire１台も持っていきます

エムネット（鹿沼市）の橋田さんも午前中からサポートしていただけるとのことで、
・モバイルルーター1台
・タブレット5台
をご持参いただけるとのことでした。

naominix, 2017-05-12:
ビスケットについてはScratch Day用のワークショップコードを先程生成しました。
もくもく会＆ハンズオン用のワークショップコード「03984405」で作成しました。
ブック１：ランド７つ
ブック２：タッチ７つ
で背景色を変えました。
ブック３：ハンズオン用１つ
で体験される方用のチュートリアルは典型的な△、海の生き物、バクバク、ランド、タッチで構成したものにしました。

mame-n, 2017-05-12:
米澤さん、
池澤さんのプロジェクターを借りるのはどうだろう。
光量の問題がクリアできるかどうかわからないけど。

星野さん
ホールには予備のプロジェクターないんですかね。

naominix, 2017-05-12:
ホールには無いようです。最初から自前でと言われました。備え付けだったら余計なセッティングいらないんですけど。

closed 2017-05-14 by vestige