TochigiRaspberryPICircle/kit #10: 木村の出展作品

akimach, 2017-05-10:
大変遅くなってしまい申し訳ありませんでしたが、出展させていただく作品を申し込みさせていただきます。

ScratchDayの趣旨にそぐわない、不適切である等のコメントがありましたら、気兼ねなくコメントをお願いします。

* RaspberryPiによる倒立振子
    * [この記事のラズパイ版](http://www.instructables.com/id/%E5%8D%8A%E6%97%A5%E3%81%A7%E4%BD%9C%E3%82%8B%E5%80%92%E7%AB%8B%E6%8C%AF%E5%AD%90/)
* Unityによるユニティちゃんの3D迷路

近日中にYouTubeでデモを公開します。

どうぞよろしくお願い致します。

akimach, 2017-05-10:
* RaspberryPiによる倒立振子

には、必要なものはありません。

* Unityによるユニティちゃんの3D迷路

に関しては一台余っているMac book Proを持参して、参加者の方々に遊んでいただく予定です。両作品を展示するには学校によくある机1つで収まると思います。

ご不便がないように念のためご報告させていただきました。よろしくお願いします。

vestige, 2017-05-10:
よろしくですー。
星野さんの予定のところにかいておきますね。

closed 2017-05-11 by akimach

akimach, 2017-05-12:
ユニティちゃんのゲームのスクショです

https://www.dropbox.com/sh/xe5fxeux5z1eqb3/AAByMw6CRzqPFiiKGWjMrcbKa?dl=0

akimach, 2017-05-12:
ラズパイはセンサーが焼けちゃったので、他のものを見繕ってもっていきます