TochigiRaspberryPICircle/kit #11: ふりかえりしたい

vestige, 2017-05-14:
とりあえずエントリーだけあげておく

naominix, 2017-05-14:
昨日は本当にお世話になりました。ありがとうございました。

準備不足なところもありご迷惑をおかけした点もあるかと思いますが、改善しながらまた実施できるようにしていきたいと考えています。

会場で制作してもらったビスケット作品の一覧はこちらです。
http://develop.viscuit.com/env/listws.html?env=11425160

ビスケットランド１－１
http://develop.viscuit.com/env/view.html?control=land:11425160:000/1-1:

ビスケットランド１－２
http://develop.viscuit.com/env/view.html?control=land:11425160:000/1-2:

ビスケットランド１－３
http://develop.viscuit.com/env/view.html?control=land:11425160:000/1-3:

ビスケットランド３－４
http://develop.viscuit.com/env/view.html?control=land:11425160:000/3-4:

ビスケットランド３－５
http://develop.viscuit.com/env/view.html?control=land:11425160:000/3-5:

Strawbeesで作ってもらった作品の写真などを撮影されていた方はいらっしゃいますか。
Creatubbles（ https://www.creatubbles.com/ ）のギャラリーを作ってシェアしたいと考えていますので、ご提供いただければ幸いです。

akimach, 2017-05-14:
みなさんお疲れ様でした。昨日は本当にいい体験をさせていただきましたよ。次回も楽しみにしておりますー。
展示させていただいたUnityの迷路ゲームは以下のレポジトリに公開しました✌️ (^○^)✌️ 
https://github.com/akimach/UnityChanMazeGame

vestige, 2017-05-27:
ラズパイ勉強会で簡単に感想を聞きました。
抜粋ですが、以下のようなことがあがりました。
ー成功できた
ー面白かった
ー刺激的だった
ーたのしかった
ー発表することでいろいろなことを考えるよいきっかけができた

よねざわとしては、次にやるとしたらいくつか課題があるような気がしています。
もう一度、折を見て確認したいと思っています。
ー会場問題
ー費用
ー進行
ー準備までの流れなど

それと、Scratch Dayとしてなのですが @mame さんが用意してくれた Scratchゲームとその横にScratchがあってよかったように思います。来客者からはScratchについて、基本的なところを聞かれる回数が多かったように思いました。Scratchイベントなのに、Scratch自体が体験できるブースがなかった気がしますね。

naominix, 2017-05-27:
色々とご感想をいただきありがとうございます。
「Scratch自体が体験できるブースがなかった」
これは本当にすみません。
実は体験できるようにNTTドコモさんからWinTabを６台ほど借りていたのですが、私の手が回らなくて準備が出来ておらず、設置することが出来ませんでした。これは完全に私が至らなかった点です。

vestige, 2017-06-15:
@naominix と 自分で”ふりかえり”した内容

- 6/15 19:00-20:30 nasu

- ざっくりと思い出すところから
  - 参加者だいたい80人ぐらい？
  - その後も結構、いろんなところから評判をきいている
  - とちぎラズパイの人たちの感想も話をした
  - 菅沼さんの発表よかった
  - 役場方面にはレポート
  - 公式サイトの方はまだ。そのうち出したい。

- 項目を分けて話をした
  - 準備
  - スタッフ
  - コンテンツ
  - 場所/時間
  - 撤収、その後

**準備**
- 期間はもう少し確保したい
- 開催を５月末の方にずらす
- 最低でも3回はオフラインでのミーティングが必要
- アナウンスを活発にした
- 情報共有のツールが必要？

**スタッフ**
- もうすこしいた方が入れ替わりとかできたかも
- 募集したい
- 教育委員会の人たちをもっと巻き込みたい
- 今回ぐらいの人数だと、ちょうどよかったかも。

**コンテンツ**
- ３ヶ月ぐらい前には決めておきたい
- 教え方、伝え方も練習しないとだめかも
- Scratchのみをやるスペース必要かも
- 感想ボード（ふせんを貼ってもらうスペース）
- メインセッション側が星野さんだけだった。メインセッションに集中できる時間を作った方がよかったかも。

**場所・時間**
- スペースはよかった
- 会場の取り方もよかった
- 車でしかこれないところ
- メインセッションの件もあり、もうすこし時間が欲しい気もする
   - が、これ以上だと、結構スタッフの人数的には厳しい

vestige, 2017-06-18:
気持ちふりかえったのでクローズ。

closed 2017-06-18 by vestige