TochigiRaspberryPICircle/kit #12: 勉強会でつぎやりたいこと

vestige, 2017-06-15:
メモしておくことがれば…

vestige, 2017-06-15:
- ADコンバーター
- トランジスタ
- くじらいさんの質問大会（いろいろ質問が山積みらしい）
- ステッピングモーターをつかった何か
- Webカメラ付き戦車
- Linuxのコアなコマンド勉強会
- ラズパイでオシロシミュレート

kotatuneco, 2017-06-15:
- Arduino
- 音声認識関係
- node-red

kotatuneco, 2017-06-18:
電子回路参考<br>
http://www.nicovideo.jp/watch/sm12618919

vestige, 2017-06-18:
戦車の話

https://www.amazon.co.jp/タミヤ-楽しい工作シリーズ-No-157-ユニバーサルプレート-2枚セット/dp/B001VZHRXG/ref=sr_1_1?ie=UTF8&qid=1497769773&sr=8-1&keywords=タミヤ+ユニバーサルプレート

https://www.amazon.co.jp/タミヤ-楽しい工作シリーズ-No-108-タンク工作基本セット-70108/dp/B002DR3H5S/ref=sr_1_1?ie=UTF8&qid=1497769911&sr=8-1&keywords=タミヤ+戦車キット

fleg949, 2017-06-18:
ラズパイで、ラジコン戦車
http://www.hiramine.com/physicalcomputing/rctank_raspi_wifi/index.html
http://rasp0035.hatenablog.com/entry/2016/11/21/100000

kotatuneco, 2017-06-18:
メモ
- タミヤ３チャンネルリモコンショベルドーザー工作基本セットx1
- バルサ材５ｍｍ厚　１０ｘ２５ｃｍ　ｘ１
- 両面テープ　or　マジックテープ
- モータドライバIC　TA7291Pｘ３
- 電池ボックス単三ｘ３本用スイッチ付きｘ２
- 単三アルカリ電池ｘ６
- 圧電スピーカーｘ１
- ブレッドボードｘ１
- ジャンパーワイヤー　メスーオスｘ９
- ジャンパーワイヤー　オスーオスｘ６
- スズメッキ直径０。６ｍｍ（カットして使用）　１ｍ程度
- 結束バンド　少々
- USBモバイルバッテリー　マイクロUSBケーブルｘ１
- ラズベリーパイ用ケースｘ１

vestige, 2017-06-18:
まとめます。次回は２部構成で

## 勉強の時間
- http://www.nicovideo.jp/watch/sm12618919
- 電子工作レシピの付録を読む

## 工作の時間
- ラズパイでラジコン戦車？を作る
- 必要なもの
  - [ユニバーサルプレート](https://www.amazon.co.jp/タミヤ-楽しい工作シリーズ-No-157-ユニバーサルプレート-2枚セット/dp/B001VZHRXG/ref=sr_1_1?ie=UTF8&qid=1497769773&sr=8-1&keywords=タミヤ+ユニバーサルプレート)

vestige, 2017-07-23:
スペーサー必要
カッター

mame-n, 2017-07-30:
ところでみなさん、hiramine.com を参考にして作業されてるのでしょうか?
欠席しておいてこんなこと言うのも失礼ですが、完成度が高いので、同じ写経(と言うのだろうか、、)ならhiramine.comがいいなと思ってしまったので。

fleg949, 2017-07-30:
自分は、まだギヤボックスすら作ってないので、変更するかもしれませんが

菅沼さんのを見本にさせてもらって、
hiramine.comにある基盤配線ではなく、ブレッドボードを使おうと考えてます。

vestige, 2017-08-05:
あとから参加する人のために、それぞれページを用意しました。
そっちにまとめていきたいです。

* [[ロボット戦車]]
* [[電気の勉強]]

fishburger, 2017-09-07:
ご無沙汰しております。鯨井です。
次回9/23(土)の勉強会、参加させていただきます。

それと、会社から下記を2セットもらったので、もしご興味があれば使っていただけるとうれしいです。
・通信モジュール（https://sakura.io/product/module_lte.html）
・ラズパイ用のHAT

mame-n, 2017-09-07:
おっ、LTEですか。どこでもラズパイですね。

https://sakura.io/product/module_lte.html

fleg949, 2017-09-23:
自分は、この配線図でやってみたいと思います。
http://qiita.com/imcuddles/items/c05cbea95db1f7469fed

vestige, 2017-09-23:
続きは #14 でお願いしますー

closed 2017-09-23 by vestige