TochigiRaspberryPICircle/kit #13: 2017/8月の勉強会の場所について

vestige, 2017-07-24:
8/19, 20がだめでした。
13日の日曜に仮予約しましたが、他の案も考えて見たいので、みんなの意見が欲しいです。

- [ ] 8/19, 20にコワーキングスペースを借りる 
- [ ] 8/13にコミュニティセンターで行う

* コワーキングスペースは以下のところにしてみようと思っています。もちろん、すこしお金がかかりますが。
  * http://socoarea.jp/HOTTAN/

7月中には決定したいとおもうのでみんなの意見をお願いします。
直接、よねざわまでメールでもよいです。

vestige, 2017-07-24:
勉強会のペースを３週に１回にしたいような気もするので、8/13でもいいとは思っていますが、ちょっとコワーキングスペースも使って見たいような...

ekzemplaro, 2017-07-25:
こんにちは。
お世話になります。
私の予定ですが、１３日と２０日は、別の用事が入っていて、１９日が空いているという状況です。
私も、コワーキングスペースに興味があります。

よろしく。
内田

kotatuneco, 2017-07-25:
私もコワーキングスペースがいいです。

fleg949, 2017-07-26:
こんにちは、高井です。

13日は、予定があるため、可能であれば、19か20日の
コアワーキングスペースが良いです。

あと、近くの駐車場ってどうでしょう？

vestige, 2017-07-29:
みてきました。1時間2700円で人数で割る感じです。ちょっと階段がキツイですが、よいところでした。3時間で予約してみようと思いますー。

vestige, 2017-07-29:
[image]

mame-n, 2017-07-29:
19,20なら大丈夫です。

vestige, 2017-08-05:
場所をきめたのでこっちはクローズ

closed 2017-08-05 by vestige