TochigiRaspberryPICircle/kit #14: next

vestige, 2017-09-23:
# 2017.09
- モーターとブレッブボードをつける

vestige, 2017-09-23:
あるといいかも？

http://akizukidenshi.com/catalog/g/gC-04351/
http://akizukidenshi.com/catalog/g/gP-02681/

vestige, 2017-09-23:
# 2017.10
- モーターとブレッブボードをつける

vestige, 2017-10-22:
* モータードライバー
* http://windvoice.hatenablog.jp/entry/2016/01/16/220541

vestige, 2017-11-19:
webiopiのインストールをした

vestige, 2017-11-19:
このページをみて、設定してみることにする。
https://github.com/TochigiRaspberryPICircle/kit/wiki/ラズパイでScratch2動かす

vestige, 2017-11-19:
次回はもう一度、戦車自体のハウジングについて見直す
はんだごても使うかも？

vestige, 2017-11-19:
菅沼さんバージョンをコミットした

https://github.com/TochigiRaspberryPICircle/kit/tree/master/tank/ctrl_webiopi

fleg949, 2018-04-26:
↓こんなキットが出てました＾＾　高っ
『親子で電子工作入門 ラズパイとスマホでラジコン戦車を作ろう！』　工作キット　ST-TANK-KIT-N 
https://www.amazon.co.jp/dp/B06XRVYFKH/ref=sspa_dk_detail_0?psc=1

akimach, 2018-04-26:
@fleg949 学生にはつらいです。実際はいくらぐらいで買えるものなのでしょうか。

vestige, 2018-04-27:
スマホもついてたら、安いかもねー

fleg949, 2018-04-28:
構成が個々で異なるので、一概には言えませんが、
2000円～　かと思いますが、皆さんいかがでしょう？

タミヤ 特別企画商品 透明 ユニバーサルプレートセット 2枚セット 69906 ￥347 
タミヤ 楽しい工作シリーズ No.100 トラック&ホイールセット (70100) ￥468
タミヤ 楽しい工作シリーズ No.168 ダブルギヤボックス 左右独立4速タイプ (70168) ￥615

あとは、ご家庭にある
小型のブレッドボードやモータードライバー
モーター駆動用の電池ケース
ラズパイ稼働用の小型モバイルバッテリ
などが必要と思います。

vestige, 2018-06-29:
ひとまず、あるところまで進んだので一度クローズします

closed 2018-06-29 by vestige