TochigiRaspberryPICircle/kit #16: 電気の勉強の進捗について

vestige, 2017-09-23:
進捗や、感想をのせていきます

メインの情報は下に

- https://github.com/TochigiRaspberryPICircle/kit/wiki/電気の勉強

vestige, 2017-09-23:
## 2017.09
* ３端子レギュレータまで勉強しました
* インピーダンスでてきた

## 2017.08
* コンデンサについて勉強した

## 2017.07
* 第２回まで進みました

mame-n, 2018-07-01:
http://usicolog.nomaki.jp/engineering/avr_lineTracer/photoReflector.html

kotatuneco, 2018-07-01:
https://www.petitmonte.com/robot/howto_photo_reflector.html

vestige, 2018-07-05:
ニコ動でつかってたフォトリフレクタは結構わかりやすい資料があった。

https://d1d2qsbl8m0m72.cloudfront.net/jp/products/databook/applinote/opto/optical_sensor/common/load_resistance_detect-j.pdf

kotatuneco, 2018-08-26:
https://qiita.com/f_nishio/items/4b9723c4e622a51aaeb5

<pre>
#!/usr/bin/env python
# Read the analog sensor value via MCP3002.

import spidev
import time
import subprocess

# open SPI device 0.0
spi = spidev.SpiDev()
spi.open(0, 0)

try:
    while True:
        resp = spi.xfer2([0x68, 0x00])
        value = (resp[0] * 256 + resp[1]) & 0x3ff
        print value
        time.sleep(1)
except KeyboardInterrupt:
    spi.close()

</pre>

vestige, 2018-08-26:
https://neuralassembly.blogspot.com/2016/06/raspberry-piscratch.html

vestige, 2018-09-30:
2018/08では、DCモーターまでやりました。
次は”内部抵抗”

kotatuneco, 2018-09-30:
タクトスイッチ回路参考
http://robocad.blog.jp/archives/678444.html

vestige, 2018-09-30:
2018/09では、
内部抵抗と、モータードライバーまでやりました。
次回はマルチバイブレーター。

ー電池の扱いは気をつけよう。特にリチウム
ーセルという単位について
ー突入電流
ーCW/CCW
ーNC

vestige, 2018-10-21:
2018/10
マルチバイブレータをやりました

実験道具？
- 小さい抵抗１k
- 大きい抵抗 5０k
- コンデンサー 10μF
- トランジスタ C1815

外部になにかつけるときは入力インピーダンスの大きいものを用意する

発信機にも使われている
- ＣＭＯＳタイマーＩＣ　ＬＭＣ５

チャタリングとスイッチの今と昔

次回はライントレーサー

vestige, 2018-10-21:
DCモーターを実験したときの回路図

[image]

vestige, 2019-03-31:
209.03
* 論理式、カルノー図までやった

vestige, 2019-04-30:
2019.04
・カルノー図をつかった実際の配線

vestige, 2019-08-27:
2019/08
- https://www.nicovideo.jp/watch/sm13939035
- マルチプレクサ・デマルチプレクサ
- むずかしかった
- MUX
- Wikiで見るとさらに混乱する

vestige, 2020-05-31:
5/31 PWM

https://www.nicovideo.jp/watch/sm14112850?mylist_sort=1&ref=mylist_s1_p1_n84&playlist=eyJpZCI6InZpZGVvX3dhdGNoX3BsYXlsaXN0X215bGlzdCIsIm15bGlzdElkIjoyMTk0NTk1MCwibXlsaXN0TmFtZSI6IumbtuOBi-OCieOBrumbu-WtkOW3peS9nCJ9