TochigiRaspberryPICircle/kit #17: Coder Dojo さくら

vestige, 2017-09-23:
# url
- https://coderdojo-sakura.github.io

#  概要
- あとでかいてね @fleg949 さん

fleg949, 2017-09-23:
CoderDojo は7〜17歳の子どもを対象にしたプログラミング道場です。
2011年にアイルランドで始まり、世界では69カ国・1,250の道場、
日本では全国に98以上の道場があります。

CoderDojoさくら
間隔：月一回
場所：JR氏家駅近辺
　　　過去2回は、氏家公民館でした。
参加費用：無料
過去の参加人数：　1回目、2回目との8名

Raspberry Pi財団、CoderDojoと合併
https://fabcross.jp/news/2017/20170531_raspberrypi_coderdojo_merge.html

fleg949, 2017-10-08:
第3回目のダイジェスト動画を、Youtubeにアップしました。
　https://www.youtube.com/watch?time_continue=3&v=Oo-DLolCjew

fleg949, 2017-11-05:
第4回目のダイジェスト動画を、Youtubeにアップしました。
https://www.youtube.com/watch?v=wQP19p1KzUU

fleg949, 2018-02-24:
第7回のダイジェスト動画を、Youtubeにアップしました。
https://www.youtube.com/watch?v=owq1mM7HL4g&feature=youtu.be