TochigiRaspberryPICircle/kit #18: プログラミングデー in 上辺見小

vestige, 2017-09-23:
# 日時
- 2017/10/28(sat) 9:45-11:45

# 場所
- 古河市立上辺見小学校 児童クラブ（東側）教室

# 内容
- ワークショップ

vestige, 2017-09-23:
まずは手伝ってくれる人はここで表明してもらえればよいのかな？

fleg949, 2017-09-23:
はい、まずは当日、来ていただける方の人数を知りたいです。

場所：
　〒306-0234 茨城県古河市上辺見１１６４　上辺見小学校

アクセス：
車　今泉地域コミュニティセンターから、新4号経由で、約53km
電車　JR古河駅下車、徒歩3km、　バス　上辺見停留所下車徒歩1分

vestige, 2017-09-23:
駐車場はあるのですか？

fleg949, 2017-09-24:
駐車場あります。
学校に数台分と、少し歩きますが、複数台止められる場所があります。

fleg949, 2017-09-24:
いまのところ、下記ワークショップと
ＳＴＥＭ教材展示を考えてます。（学校側と調整中）

■ワークショップ（体験会）
・タブレットを使ってプログラミングを体験してみよう
	　https://code.org/minecraft
　　　Hour of Code　https://code.org/learn
・パソコンやタブレットを使わずに、体験してみよう
	ＣＳアンプラグド：ルビィのぼうけん、テラと７人の賢者
・ボール型ロボットを使ってプログラミングを体験してみよう
	Sphero SPAK+	

■ＳＴＥＭ教材
・Sphero SPARK+
・mBot
・GLICODE　ほか

fishburger, 2017-09-30:
今日、高田さんからこちらのお話しをききました。
上辺見小学校以外の子供の参加も大丈夫でしょうか。
大丈夫でしたらうちの子供もつれて参加（お手伝い）させてください。

vestige, 2017-09-30:
大丈夫ですよー

vestige, 2017-09-30:
て、僕が答えてもよかったんだろうか笑
来週中にコンパスでスタッフの人数確認用のサイトを用意しますね。

fishburger, 2017-09-30:
ありがとうございます。

fleg949, 2017-09-30:
大丈夫ですよ～
出来れば、来週水曜日までに、だいたいの人数
を把握したいですが、 いかがでしょうか？
今のところ、四名さんぐらいですかね

vestige, 2017-10-02:
https://torpc.connpass.com/event/68479/

vestige, 2017-10-02:
とりあえず、上に登録してもらうと連絡しやすいかもー

mame-n, 2017-10-04:
せっかくの機会ですけど、都合がつけられません。
またぜひ、お声をかけてください!!

naominix, 2017-10-04:
応援に行きたいですが、スケジュールが重なっているため行けないので、また別の機会がありましたらその時にお願いします。

fleg949, 2017-10-04:
mame-nさん、naominixさん　
ご連絡ありがとうございます。お気持ちだけでもとてもうれしいです。

kotatuneco, 2017-10-17:
fleg949 さん、私の知り合いでぜひ参加（お手伝い）したい方がいるのですが、よろしいでしょうか？

vestige, 2017-10-17:
@fleg949 @kotatuneco たぶん大丈夫だと思いますよ。連絡とかが周知することがあるかもしれないので、上のコンパスから登録だけはしてもらった方がよいかもー。

kotatuneco, 2017-10-17:
@vestige コンパスに登録するよう伝えておきます。

fleg949, 2017-10-17:
ありがとうございます。

資料を、後ほど送ります。

oogakim, 2017-10-24:
@vestige @fleg949 10/28行けそうなのでconnpassには登録しました。
当日の集合時間と、それまでに何か準備できること教えて欲しいです。
何ができるかわからないので見学程度になりそうですけど。

fleg949, 2017-10-24:
OGAKIさん
ありがとうございます。

先程Facebookのほうへ、情報を送りました。

わからないことあれば、遠慮なく問い合わせください。

/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/
高井 和之
E-Mail ： fleg949@gmail.com
携帯電話 ： 080-8912-6621
/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/

fleg949, 2017-10-28:
今日は、遠くまで来ていただき、
本当にありがとうございました。

忘れ物が、いくつかありましたので、
次回お会いしたときお渡しします。

あと、バタバタしていて、
写真をとるのを忘れて
しまったんですが、
どなたか、子供たちの写真あります
でしょうか？

vestige, 2017-11-05:
無事におわったのでクローズ

記事のリンクを載せておきます。

http://ict-enews.net/2017/10/31kamihemi/

closed 2017-11-05 by vestige