TochigiRaspberryPICircle/kit #19: 宇都宮大学での子供向けイベント

vestige, 2017-09-23:
# タイトル
- まだない？

# 概要
- @akimach が主催予定

akimach, 2017-09-24:
@TochigiRaspberryPICircle/tropic 
とりあえず開催できる体制と場所はあるので、何かやりたいなーというのが今の段階です。これからまとめて行きたいと思います。

akimach, 2017-09-24:
# TensorFlow勉強会 2017年9月24日での話し合い

プログラミング教育の百花繚乱ぶりを体感した。

* [GLICODE®](http://cp.glico.jp/glicode/)
* [コンピュータサイエンスアンプラグド](http://csunplugged.jp/)
* [Micro:Bit](http://microbit.org/ja/)
* [MinecraftのHour of Codeチュートリアル](https://code.org/minecraft]
* 坂村健によるコンピュータを使わないコンピュータ教育（トランプとか使う）
* [mBot](https://www.kenis.co.jp/mbot/)
* etc

「教材やデバイスどうするか問題」で上がった声

* ラズパイだとOSのセットアップが手間
* Arduinoだとそういう手間がない
* micro:Bitもラクらしい
* コードをかかせるのか？
* やはりScratchか

vestige, 2017-12-24:
1月には具体的な話をはじめたい。

akimach, 2017-12-29:
@vestige 
分かりました。9月の開催を目指したいと思っています。

akimach, 2018-06-02:
#33 で話すので閉めます。

closed 2018-06-02 by akimach