TochigiRaspberryPICircle/kit #25: 那須町開催の2つのイベント

mame-n, 2017-12-19:
那須町で2つ、ICT教育関連のイベントがあり、主催の星野さん(那須塩原クリエイティブラボ)からお誘いをいただきました。

1つ目は1月13日(土)に、那須町の東陽小学校で開催されます。以下、星野さんから。「東陽小はこれまでのプログラミングデイとは異なり、土曜学校の日に当てている関係で、参加が全児童対象になっているのです。そのため、ブースがたくさんないと体験するのに行列が出来る可能性もあり、出展者が多いほど良いのではないか、と考えています。東陽小は元々の東陽中学校の校舎に移転したばかりなのですが、元が中学校なので教室も空きが複数あるので、そこを出展スペースとして使うと４，５教室くらい使える状況になっています。」
もう1つは2月3日(土)に、那須町文化センターで開催されます。ここは5月にScratchDay in 那須町が開催された場所です。「2/3は那須町文化センターで大・小ホールを午前中使える状況で、こちらも企業出展や佐野日大高校安藤先生によるドローンプログラミングなども実施予定です。」安藤先生は星野さんと栃木に2人のMicrosoft認定の教育イノベーターです。https://sway.com/8QuWXR9lbwAcvPOz?ref=Link&loc=play

ScratchDay in 那須町に参加されたことがある方は、それに似たイベントと思って良さそうです。
出展ブースで展示したい方や、ICTガジェットで子供達に教えたり、一緒に遊びたい方、是非、参加しませんか。ガジェットは星野さんが持っているものなら借りることは可能と思います。(残念ながらプレゼンできる場はなさそうに見えます)

vestige, 2017-12-20:
1/13 用のをたてました。
https://torpc.connpass.com/event/75135/
お手伝いできる人はまずは登録してくださいー

mame-n, 2017-12-25:
1/13のやや詳細な情報です。
http://canvas.ws/workshop/16671

時間は9時から12時くらいは展示することになりそうです。
短時間ですが、自作品展示、ICTグッズを子供や親に教えたりできそうですので、ぜひぜひ、ご参加ください。

Viscuitの作者の原田ハカセの講演を聞くだけでも楽しいかもしれません。
(その場合は上記のページから申し込んでください)

naominix, 2017-12-27:
お世話になります。
出展場所ですが、東陽小学校校舎３Ｆの空き教室が４，５教室あるので、そちらを使うことになる見込みです。

出展企業さんは１／１３・２／３ともに、内田洋行、アーテック、ナリカ（展示のみ）、NTTドコモの４社です。

内容的には

・ルビィの冒険体験コーナー
・LEGO EV3、WeDo体験（タブレットでプログラミング）
・文教向けタブレット、電子黒板、その他普通教室で使用するＩＣＴ機器実践例展示
・プログラミング教材と理科実験ＩＣＴ機器展示（LEGO、教育用ICT計測システム（イージーセンス）紹介、理科実験全天球カメラ活用等）
・Artecロボを使った信号機、モーターカーのプログラミング体験
・embot( https://www.embot.jp/ )プログラミング体験・展示

が予定されています。

那須ラボからはOsmoやSphero SPRK+やOzobotやMeetMiPやmicro:bit複数台などお貸しすることが可能なので、これを使いたいというリクエストがありましたらお知らせください。

東陽小の方はプレゼンタイムはありませんが、文化センターの方はプレゼンタイムを大ホールで実施予定です。大人も参加可です。１時間程度しか時間がないので、登壇人数によってはＬＴ的になるのではないかと思います。

２／３は午前中展示体験が小ホールで９：３０ー１１：３０、プレゼンタイムが大ホールで１０：３０ー１１：３０、１１：３０ー１２：００片付けとなります。

午後は１：３０から佐野日本大学高等学校安藤先生によるドローンプログラミング＆飛行体験会があります。大・小ホールは使えませんが、会議室や無線ＬＡＮが使える視聴覚室は使用可能なので、その１室を使ってドローンプログラミングを実施予定です。

yakitori55, 2017-12-27:
藤原でなくて、yakitori55です。
那須のイベント、参加させていただきますので、よろしくお願いします。

vestige, 2017-12-27:
@yakitori55 連絡することがあるかもしれないので、  https://torpc.connpass.com/event/75135/ 登録お願いしますね。

inamihiroyuki, 2017-12-27:
1/13お手伝いしますーー。（勝手に遊んでるかも）

fleg949, 2017-12-28:
1月13日
　mBot x 2台（青、ピンク）、Sphero SPRK（＋無し）を持っていこうと思います。
　デモするスペースとかありますでしょうか？
　那須ラボさんのSPRK+と合わせて、何かデモできるかな～
　あと、どなたか、デモ用に、タブレット（Android or iOS）を、貸してもらえると
　うれしいです。

fleg949, 2017-12-28:
-

naominix, 2017-12-28:
@fleg949 さん、デモ用タブレットにはSphero Eduアプリが入っていればよろしいでしょうか。那須町保有のプログラミング用iPadが20台程ありますので、そちらを使用することも可能です。

那須ラボにはSPRK（＋無し）もあります（バッテリーがヘタってるかな？）。

展示場所の教室は１部屋丸々使えますので、必要であればSpheroのジャンプ台とかも必要であれば持っていくことが可能です。

fleg949, 2017-12-29:
@naominix さん
>Sphero Eduアプリが入っていればよろしいでしょうか
　はい、Spheroのデモは、それでお願いします。
　あと、１台でよいので、mBlockアプリが入っているのが
　あるとうれしいです。
[http://www.mblock.cc/software/mblockapp/](url)

mame-n, 2018-01-06:
星野さん、
1/13ですが、ミルククラウンのような展示を那須ラボブースの範囲で行いたいのですが、可能でしょうか。
またScratchDayの時のようなViscuitワールドを行う予定はありますか?

当日ですが、私は8時過ぎに現地着を考えています。
先生は何時ころに現地入りの予定が教えてください。

naominix, 2018-01-06:
当日ですが、８時過ぎだと少し早いかもしれません（学校を開ける時間がもう少しあとになる見込みです）。
私と指導主事は前日にある程度の現地の準備を行って、当日は９時頃に現地にという予定でおります。
機材のセッティングで私の方でできることがあれば前日に実施してみたいと思いますがいかがでしょうか。

その前提で、ミルククラウンの展示は可能ですが、私の方で実施できることをご指示いただければと思います。

Viscuitワールド（ランド）、もちろん実施は可能なのですが、どなたかそこにサポート入っていただく必要がありますが、可能でしょうか（３Ｆで機材を運ぶのが大変なのでプロジェクターで投影するような感じが良いかもしれませんね）。

講演がはじまってしまうとそちらの方に入って記録をつきっきりでやることになってしまいます。

kotatuneco, 2018-01-07:
星野さん

とちぎラズバイ同好会の高田と申します。
ミルククラウンの展示ついて連絡させていただきます。
（展示については私が中内さんに質問した内容です）

ミルククラウン撮影の展示は、機材が多く、
事前に送ることが難しいと考えています。
（私の作りの問題なのですが、）
また、遮光等に影響するため、
機材のセッティングには時間がかかってしまうため、
今回の展示には、厳しいと考えています。

せっかく、展示の件、ご検討いただいたのですが、
今回は、別のお手伝いとして参加させて下さい。

セッティング不要な倒立振子のような
電子工作であれば持って行くことはできますので、
１／１３に関わらず、こういったイベントに
声をかけていただければ幸いです。

以上、宜しくお願いします。

naominix, 2018-01-09:
高田さん、お世話になります。

ミルククラウン展示の件、了解いたしました。
倒立振子の電子工作はとても良い事例として見てもらえると思いますのでぜひお願いいたします。

東陽小の方での展示は2/3のプレ開催という位置づけなのですが、土曜学校の日になっている関係で集まる児童数がこれまでのプログラミングデイに比べると非常に多いということもあり展示を充実させたい反面、短時間で撤収となるなど制約も多い状況になっています。

2/3は那須町文化センター小ホールでの展示が可能ですので、そちらではぜひミルククラウンの展示をお願いできれば幸いです。

皆さまのご協力に深く感謝いたします。
今年もScratchDay in Nasuの方も開催出来るように努力いたします。

naominix, 2018-01-09:
あと、那須ラボから

Osmo ベースキット２セット

の貸し出しも可能です。
Osmo CodingとCoding Jamをそれぞれやることは可能かと思います。

ラボが保有しているiPad(5G)とiPad mini2も使えますので必要に応じてリクエストください。

vestige, 2018-01-09:
スタッフ向けにすこし情報をまとめました。
https://torpc.connpass.com/event/75135/

kotatuneco, 2018-01-10:
星野さん
倒立振子の件、了解しました。
持参させていただきます。

fleg949, 2018-01-12:
東陽小学校は、那須中学校に移動したそうです。

GoogleMapで、検索して来ると、
今は使われていない小学校に来てしまいます。

mame-n, 2018-01-14:
1月13日はお疲れ様でした。
子供たちの数に圧倒されましたが、盛況のうちに開催できたと思いました。

次は2月3日に開催予定の、同じくプログラミングデーin那須町です。
ScratchDayと同じ場所で行います。
近々、星野さんと打ち合わせを持ち、プログラムを具体化したいと思っています。
現在のところは下記の2つになりそうです。
1、1/13同様に、展示参加、もしくは展示サポート。ガジェットを使っての子供へのアドバイス等々。
2、5分〜10分の発表 (ご存知なら、ライトニングトークス形式になりそうです)
興味のある方は2月にもぜひぜひご参加ください。よろしくお願いします。

詳細は追ってご連絡します。

vestige, 2018-01-14:
2/3用のをたてました。
https://torpc.connpass.com/event/76820/

mame-n, 2018-01-14:
ありがとう

mame-n, 2018-01-20:
2月3日のプログラミングデーin那須町8 ファイナルの件です。

https://www.dropbox.com/s/9xxc2hc3dwz4vde/20180203_プログラミングデイ那須町8回目_20180118.pdf?dl=0

今回は一般発表の場が設けられます。ScratchDayを開催したホールを覚えている方は、あの場所で展示を実施し、発表は大ホールで行います。

https://www.town.nasu.lg.jp/0227/info-0000000679-1.html

1/13のプログラミングデーやScratchDayのように展示をお手伝いしていただける方を募集します。また展示だけでなく、発表者として自分が子供の頃、若かりし頃(まだ若い方は除く、、、)にどのように技術に接してきたか、学んできたかを子供たちに伝えたり、最近の技術成果を見せる方での参加も募集します。
発表は一人5分〜10分程度です。
準備期間が短いですが、ぜひぜひ、ご参加ください。

ちらしでわかるとは思いますが、午前中に展示と発表会があり、午後はドローンのセッションのみとなります。午後も見学なら参加可能なので、時間に余裕があれば午後までご参加ください。

naominix, 2018-01-21:
那須町文化センター小ホールの舞台図になります。

https://www.dropbox.com/s/g4ybn4f96h72lgx/%E5%B0%8F%E3%83%9B%E3%83%BC%E3%83%AB.JPG?dl=0

長テーブル・椅子などを倉庫から出して利用します。

naominix, 2018-01-23:
以下、展示・体験の主な内容です（一部変更があるかもしれませんが、現時点での内容です）

■企業・団体出展（那須町文化センター小ホール）■
各種ＩＣＴ／プログラミング教材等体験ブース
展示体験時間帯：０９：００－１２：３０（１３：００完全撤収）

◆出展企業
・株式会社アーテック
　　　－アーテックロボ教材体験
・株式会社内田洋行
　　　－レゴマインドストームＥＶ３／WeDo2.0教材体験
　　　－MESHプログラミング
　　　－ルビーの冒険アンプラグドＣＳ体験
・ＮＴＴドコモ
　　　－embotプログラミング体験
　　　－はなして翻訳アプリ体験
・株式会社ナリカ（展示担当：有限会社イナバ）
　　　－理科関連教材体験
・藤井産業株式会社
　　　－little bits教材体験
　　　－MetaMoji Classroom展示
　　　ー高性能電子黒板展示

◆出展団体
・ＮＰＯ法人CANVAS
　　　ーHour of Code体験
　　　ーmicro:bitプログラミング体験
　　　－Minecraft: Education Edition ✕ MakeCodeプログラミング体験
・とちぎRaspberry Pi同好会（一部機材提供：那須塩原クリエイティブ・ラボ）
　　　ーミルククラウン撮影システム展示
　　　ーOsmo Coding/Coding Jamプログラミング体験
　　　ーラズパイロボ展示
　　　ーストロビーズ
　　　ーワミー
　　　ーその他

naominix, 2018-01-27:
改めて文化センター関連の資料をシェアいたします。
写真を入れてあります（ScratchDay in Nasuの準備風景写真も）

・文化センター見取り図、小ホール写真等
https://www.dropbox.com/s/2ikr6gedojgladh/%E3%83%97%E3%83%AD%E3%82%B0%E3%83%A9%E3%83%9F%E3%83%B3%E3%82%B0%E3%83%87%E3%82%A4%EF%BC%98%E5%B0%8F%E3%83%9B%E3%83%BC%E3%83%AB%E6%A6%82%E8%A6%81.pdf?dl=0

fleg949, 2018-01-29:
前回と同じく、mBot x 2台（青、ピンク）を持っていこうと思います。
デモするスペースとかありますでしょうか？

あと、どなたか、mBotのデモ用に、タブレット（Android or iOS）を、
1台貸してください<m(_　_)m>。

naominix, 2018-01-29:
@fleg949 Takaiさんありがとうございます。
基本的には各ブースは壁面にそって並ぶことになりますので、デモスペースを想定したテーブル配置などで工夫していただくことになるかもしれません（テーブルと壁面のスペースなど）

おおよそ、ブースの配置を週中頃までに確定させたいと思います。

タブレットに関しては那須町のプログラミング用iPadを20台用意する予定ですので、そこにmBotアプリをインストールして対応できればと考えております。

那須町保有のSphero✕10が先週末に届きましたので、これを活用することも可能かと思います。

kotatuneco, 2018-01-31:
タカタです。
２／３ですが、本業がバタバタしており、
仕事になってしまいました。
ギリギリでの報告になり、申し訳ありません。

to:藤原さん
　ミルククラウンの件は、明日、相談させてください。

naominix, 2018-01-31:
プログラミングデイin那須町８当日案内
https://www.dropbox.com/s/8zvuckqjlfq78fp/%E3%83%97%E3%83%AD%E3%82%B0%E3%83%A9%E3%83%9F%E3%83%B3%E3%82%B0%E3%83%87%E3%82%A4%EF%BC%98%E5%BD%93%E6%97%A5%E6%A1%88%E5%86%85.pdf?dl=0

当日の配置を決めてみました。
実際には前日の準備段階で調整もしたいと思います。

naominix, 2018-01-31:
文化センターそのものの開場が９時ピッタリということなので、実質的に準備しながらの展示スタートとなります。
今回は東陽小のときとは異なり、いきなり大量の人が待っている、という状態にはならない（ハズ）ですので、準備しながら展示・体験をしていければと思います。
前日に事前準備をしているところもありますので、早く来た方はそちらから順を追って回ってもらうようにしたいと思います。

kotatuneco, 2018-02-01:
タカタです。
準備等で、迷惑がかかりますので、
ミルククラウン展示は辞退させていたきます。
直前で申し訳ありません。

藤原さん、私の分まで、
お手伝い、よろしくお願いします。

mame-n, 2018-02-02:
では集合は8:50にしましょう。9時になったらすぐに会場に入れるように。
星野さん、展示品は前回同様と思っておいていいですか？
高井さん、タブレットを星野さんからと言ってましたが、大丈夫そうですか？

naominix, 2018-02-02:
本日、小ホールの事前準備に行ってきました（大ホールも）。
配置図のあたりに３つほどテーブルを出し、椅子は２つ程度だけ出してあります。明日の朝、追加で出したいと思います（時間が無くて十分な数出せませんでした）。

今回の機材ですが、

・那須町学校教育課保有Sphero SPRK+10台（那須町学校教育課プログラミング用iPad空き端末を適宜使用）
→ジャンプ台なども会場に置いてあります。iPadにはSphero Edu導入済みです。Tickleはアカウント未登録なので一部ブロックが使えない状態です
・Osmo Coding／Osmo Coding Jam（那須町学校教育課プログラミング用iPad18,19）
・ストロービーズ／ワミー
→テーブルの上に置いてあります。

mBot用にアプリをインストールしてあるのが那須町学校教育課プログラミング用iPad20だったハズ（すみません。明日確認すればすぐに判別できます）

なお、那須町学校教育課プログラミング用iPad7，8台は充電が不十分な状態である可能性が高いです（本日夕方まで黒田原小学校で使用していたためで、明日朝一で充電開始します）

mame-n, 2018-02-03:
みなさん、本日はお疲れ様でした。
最初はメーカー展示に子供達を取られてしまいましたが、後半は多くの供達に楽しんでもらえました。
ありがとうございました。

Viscuitランドですが、https://develop.viscuit.com/env/school.html の
「授業に参加するこちらからコードを入力してください」で 70242717 を入力すると見られます。
上から5つ目の濃い青の円をクリックし、(鉛筆)(点々)(四角)のうち、四角を押すとランドが表示されます。

もしかしたら直に行けるかも。
https://develop.viscuit.com/env/view.html?control=book:05603767:d.json:0:&savepath=70/24/27/17

naominix, 2018-02-03:
皆さま、プログラミングデイin那須町８は大変お世話になりました。
Sphero、mBot、Osmo Coding、ストロービーズ、ワミーなどさまざまな取り組みをしていただき感謝いたします。楽しく、真剣に取り組んでいる子どもたちの様子を見ることができました。

@mame-n さん、ビスケットランド、素敵でした。ファシリテーターとして大活躍！ですね。

CoderDojoさくらの方も再びお越しいただき本当にうれしかったです。

那須町内の小学校をめぐって実施してきた2017年度プログラミングデイはこれで終了です。
これをきっかけにして「毎日がプログラミングデイ」となるような取り組みが各学校、地域で継続的に行われるようにするために、裾野を広げ、保護者の理解を広める活動も続けていきたいと思います。

ご協力いただいた皆さんの地域でもお手伝い出来ることがあればお声掛けください。
本当にありがとうございました。

closed 2018-03-01 by vestige