TochigiRaspberryPICircle/kit #26: Scratch Day in NASU 2018について

vestige, 2018-01-29:
* 2018/5にやるよ、ぐらい。すこしずつ詳細つめます。

vestige, 2018-02-25:
* やってみたいこと

- ロボット戦車の展示
- ScratchでLチカ
- はんだごて演習
   - ブレッブボードと、リアルはんだで違いを確認する

vestige, 2018-03-01:
次にやることは
* ひとつずつissueを立てる
* 担当を決める

vestige, 2018-03-18:
Lチカはやってみた
- https://github.com/TochigiRaspberryPICircle/kit/issues/30

vestige, 2018-03-18:
4月は
- 日程を決める
- ６月開催しないときは、単独イベントを考えてみる。きむらくんとも話してみる
-  はんだごてブースについて考えてみる

naominix, 2018-03-24:
遅くなりすみません。
https://day.scratch.mit.edu/events/5749/
6/2に昨年と同じ文化センター小ホールを予約しました。
終日空いているとのことでしたので、夕方まで借りる方向で予約しました。

naominix, 2018-03-30:
那須町文化センターの方、正式に利用申請の方を提出しましたのでご報告します。

naominix, 2018-03-30:
直接関係はありませんが、Scratch Day 2018 in Tokyoは６月３日（日）に日程が変更になったようです。

mame-n, 2018-03-31:
6月2日までRuby会議が仙台で開催されます。
中内はScratchDay in那須町に参加します。

vestige, 2018-04-01:
よねざわです。
ぼくはRuby会議にでているので参加できませんが、ブースなどについては、勉強会の中ではなしておきますね。

mame-n, 2018-05-02:
当日のパンフレットが手に入りました。リンク先にpdfあります。
開始時刻は12:30です。
15:30までの前半に展示、15:30から発表が組まれています。

集合時刻は11時にしましょうか。
それと私が那須ラボ星野さんからお借りする予定のガジェットは後ほどリストアップします。

[ScratchDayinNasu2018.pdf](https://github.com/TochigiRaspberryPICircle/kit/files/1965872/ScratchDayinNasu2018.pdf)

mame-n, 2018-05-28:
ScratchDay2018でスヌーズレン(Snoezelen)というものを展示してみることになりました。
設置を手伝っていただける方、ぜひぜひ、お願いします。
正直なところ、うまくできる自信はないのですが、とにかく挑戦してみようと思っています。

スヌーズレンとは、
https://www.shimada-ryoiku.or.jp/tama/intro/snoezelen.html

下手くそだけどイメージスケッチ
[https://drive.google.com/open?id=1M10-YC_G5UcxJMPcPkH5GPQCOaBtZbtz](url)

まだまだ勉強中ですが、何か1つでもこれをやることで気づけたらいいなと思っています。
みなさん、よろしければ作成を一緒にいかがですか。

naominix, 2018-05-30:
◆タイムテーブルについて◆
昨日公開したタイムテーブルはチラシのタイムテーブルと３０分ほどズレてしまいました。
このあたりは当日の状況を見ながら前後させて調整したいと思います。

◆使用ガジェットについて◆
使用したガジェットについてご希望がありましたらお知らせください。
毎回持っていっている定番ガジェットは今回も持っていく予定です。
新しく追加するのはトイドローン系になります（あまり台数が無いのとバッテリーがすぐ無くなります。また、安全に配慮し、必ず大人がついている状態で飛ばすことが条件としたいと思います）

mame-n, 2018-05-31:
星野さん、ありがとうございます。
明日の私の行動です。
8:05 星野さん宅 (荷物を車に積む)
8:20 星野さん宅出発
8:55 那須町文化センター着
9:00 設営開始
11:30 スタンバイ状態で食事
12:30 開演

mame-n, 2018-05-31:
ホールでのガジェット
- Sphero
- OSMO

スヌーズレン
- 球体モニター+MacBook
- プロジェクター
- ミラーボール
- Skoog+iPad mini
- クッション (星野先生、ぜひぜひお借りしたいのですが、、)
- プラネタリウム
- スピーカー+iPhone
- LEDイルミネーション
- テーブルタップ(3)

naominix, 2018-06-01:
那須町文化センターに到着して会場を開けたら、一度那須町庁舎に行かせてください。
プロジェクターなどいくつか町教委に機材を置いてある状況なのでそちらから運びたいです。

トイドローンはDRONE STAR 01を２機用意します。５分もしない内にバッテリー切れで飛べなくなるので充電ケーブルとUSB充電器を持っていきます。ダンボールなどの着陸ポイントにうまく飛ばせるか飛行体験などをしてもらえればと思います。

もう１つのトイドローンTELLOは集合写真空撮で使おうかと思っています（バッテリーは予備が２つあります）。

Spheroは那須町から10台。ラボから１台。

OSMOは、CodingとCoding Jamに絞って持っていきます（組み合わせるiPadが問題カモ）。

球体モニター（World EYE）は準備しました。

ミラーボールも準備しました。

Skoog＋iPad miniも準備しました（動作確認済み）。

クッションというのはYogiboのことでしょうか？車に乗るだろうか・・・。

私の方からはこれくらいかな。

vestige, 2018-06-07:
無事に終わったようなのでクローズします。

closed 2018-06-07 by vestige