TochigiRaspberryPICircle/kit #33: 夏休みこどもプログラミング教室

akimach, 2018-06-02:
この前、 @vestige さんと話した内容です。

* 候補日：8/5?、~9/15, 9/22~
* 定員：20名（=親子10組程度）
* 内容：プログラミングのワークショップ
  * ScratchとRasPi
  * 栃Piメンバーによる発表会・デモ
* 時間：13:00 〜 16:00
* 対象：小学生全学年とその保護者
* 会場：宇都宮大学
* 備考：
  * 宇都宮市の学校に広報予定
  * PCは10台程度貸出可能
  * 学生は3、4人程度お手伝い予定

akimach, 2018-06-05:
以上の内容で、昨日に提出しました。

vestige, 2018-06-07:
ありがとう。次の勉強会で説明しようと思います。
ていうか、8/5だとやっぱりあまり時間なさそうだねー^^;

akimach, 2018-06-07:
@vestige 
@vestige 

> 普段の活動報告、またはこの勉強会を通してやりたいことをこの場で発表しあうようなことをしたいです。

はい。前回お話したとおり、プログラミング教室の構成や流れについては、よねざわさんはじめとちぎラズパイ勉強会のみなさんにお任せしたいと思います。

akimach, 2018-06-07:
> #30 をベースにすこし掘り下げられたら

PCは貸出できます。ラズパイはないので難しいですね

akimach, 2018-06-08:
@vestige 

聞いてみますが、ほぼ確定だと思います。

akimach, 2018-06-08:
唐突なご提案でわちゃわちゃしてしまい、申し訳ございません。今後、冬休み、春休みなどにも開けたらなーと思うので前向きに検討していただけるとありがたいです。

負担を軽くするためにいくつか策は施せると思うので、ぜひ言ってください。

* 前半は僕を含め学生でプログラミング教室の構成を考えて実施、後半にラズパイ勉強会のみなさんが発表会
* もう少し時間を短くして、シンプルにする

などなど

いずれにせよ話し合わないといけませんね。

fleg949, 2018-06-08:
もしよければですが、
さくらや近隣のCoderDojoに、
声かけてみて良いですか

先日ScratchDay in NASUのように
都合が合えば、水戸やひたちなかから
も来てくれるかもです。

akimach, 2018-06-08:
@fleg949 @vestige 

> さくらや近隣のCoderDojoに、
> 声かけてみて良いですか

ありがとうございます。ココらへんで話は留めておいて、よねざわさんがおっしゃるようにまずは次回のラズパイ勉強会で話し合ったほうがよいかと思います。会場の規模的にも大規模にやることは難しいですし

vestige, 2018-06-13:
今日、 @mame-n さんと話した。話した中から自分のメモをのこしておく

　
- すでにあるコンテンツをやるのはよい
- はじめに完成品を見せる。お絵かきを自由に
- 子供中心、大人中心？
- ラズパイ はいくつ用意できるか？
- キーボード、マウス、モニターも必要。用意できるか？
- 感想（ワールドカフェ）は子供が飽きてしまいそう
- ２ターンにして、ロボットアンテナからなにか追加できないか？センサーっぽいのとか？

akimach, 2018-06-13:
@vestige 

ありがとうございます。確認してみます。

> 9月ぐらいでもいいのでは？夏休みの土曜日は予定合わせにくい？

附属小学校などにも連絡はしますし、これは広報次第だと思います。

akimach, 2018-06-13:
いろいろとアイディアはあって、考えうる変数が多くなってしまうので

* 8/5開催
* mac book pro 10台程度貸出（マウスもおそらく大丈夫です）
* 数名学生のお手伝い可能

という制約条件のなかで、まずやれそうなことを探れたらなーと思ってます。

vestige, 2018-06-29:
https://github.com/TochigiRaspberryPICircle/kit/wiki/programmingEventTorpc

vestige, 2018-07-01:
きになること（7/1）

- wifiはどうなる？
- OSを揃えたSDカードを用意した方がいい？
- モニター、キーボード、マウスはあった方がいい
- 配線のケーブル、抵抗

vestige, 2018-07-01:
https://torpc.connpass.com/event/93290/

akimach, 2018-07-01:
> wifiはどうなる？

用意できない可能性がありますが、確認してみます。

> OSを揃えたSDカードを用意した方がいい？

Rasbianのバージョンで差はないと思うので、最近のものであれば大丈夫ではないかと思われます。

> モニター、キーボード、マウスはあった方がいい

米澤さんに個別に連絡します。

> 配線のケーブル、抵抗

こちらでご用意するのは難しいと思います。

vestige, 2018-07-02:
スタッフだけでまずはいくつラズベリーパイあるか確認だね。
GUI使うのでラズパイ3がいいような気がする。
2のB+でも大丈夫かなぁ？

kotatuneco, 2018-07-02:
貸し出せる機材について連絡します。
・Pi3 x 2台
・Pi3B+ x 1台
・PC用ディスプレイ x 2台

vestige, 2018-07-05:
@kotatuneco ありがとうーございますー。

akimach, 2018-07-07:
ご用意できる機材：

・Pi3B+ x 1台
・PC用ディスプレイ x 3台以上
・マウス x 3台
・キーボード x 3台

vestige, 2018-07-13:
@akimach @fleg949 ありがとうございます。

vestige, 2018-07-17:
2ターン目をタクトスイッチでと考えていたけど、よく考えたらブレッブボード使わない予定だった。
センサーは困りそうだよね、どうしようかなぁ。

mame-n, 2018-07-18:
予習したいのですが、やろうとしているロボットの参考Webとかありますか?
タクトもそうですが、試しにLEDやる場合もブレッドボード便利ですよね。かき集めますか。
でもブレッドなしのLEDだとワニ口クリップとか使おうと思っていた?

vestige, 2018-07-19:
@mame-n 
https://projects.raspberrypi.org/en/projects/robot-antenna
参考にしようとしてるのは上のサイトです。
メス-メスのジャンパーワイヤーは用意してありますよ

vestige, 2018-07-19:
あと、参加者が定員になったそうですー。
なので、機材とかを再確認しないとだ

vestige, 2018-07-19:
一応、現状の持ち物をまとめて見た。ちょっと機材足りない。どうしようか。

https://github.com/TochigiRaspberryPICircle/kit/wiki/programmingEventTorpc#用意するもの

naominix, 2018-07-19:
PiServer
https://tech.nikkeibp.co.jp/atcl/nxt/column/18/00079/00004/

でThinクライアント運用できるでしょうか。
若干不安もありますが・・・

vestige, 2018-07-19:
@naominix ありがとうござまいます。もしかすると、お声掛けさせてもらうかもです。ThinClientもあるんですね。まぁありますよね。GPIOを使うのでこっちはちょっとだめそうかもですー。

akimach, 2018-07-19:
[続き](https://github.com/TochigiRaspberryPICircle/kit/issues/33#issuecomment-406230343)：

* Pi3：1台追加

ボランティアとして参加予定の学生のものです。

rohatodenirou, 2018-07-20:
私もお手伝いします。

・Pi3 x 2台
・PC用ディスプレイ x 2台
・マウス x 2台
・キーボード x 2台

vestige, 2018-07-21:
@rohatodenirou ありがとうございます！
ディスプレイ、キーボード、マウスは大丈夫そうなのですが、ラズパイ の電源、HDMIのケーブルがたりなそうなのでその辺を記述してもらえるとうれしいですー

それと、２ターン目のをつくってみました。このぐらいなら、どうかな？
https://scratch.mit.edu/projects/237011851/

vestige, 2018-07-21:
りんご動かしてみるバージョン
https://scratch.mit.edu/projects/237012531/

vestige, 2018-07-21:
mameさんのもはっておくー
https://scratch.mit.edu/projects/237022181/

mame-n, 2018-07-22:
星野さん
文面が寝ぼけていました。

ラズパイをお借りする件ですが、やはりこちらで十分に用意できないので3台ほどお借りしたいと考えています。
せっかくですからB+をお願いできますか。
23日の受講の時に妻に渡していただくのが都合がいいのですが、いかがでしょう。
ちなみにお借りするラズパイ用の電源ケーブルはお持ちでしょうか。あればそれもお借りしたいのですが。

米澤さん、昨日の話では3台借りると記憶していますが正しいですか。

vestige, 2018-07-22:
@mame-n 昨日、星野さんに連絡しておきました。ごめんなさい。一緒にいれておけばよかった。３台借りる話はしてありますよ。

naominix, 2018-07-22:
@mame-n さん、3B+3台用意しておきます。電源（3A出力×２、2A出力×１ですが、月曜日までに新しい3A出力が届けば全部3Aになります）も３つ用意します。

mame-n, 2018-07-22:
了解です。では23日、お願いします。

rohatodenirou, 2018-07-22:
下記用意できます。
HDMIケーブル x 2
3A電源 x 2
2.5A電源 x 2
coderdojo sakuraで使ってるwifi(42台同時接続可能)も借りられると思いますが、どうしますか？

vestige, 2018-07-22:
@rohatodenirou  wifiはいまのところ必要ないので大丈夫ですよ。ありがとうございます。

vestige, 2018-07-22:
HDMIで接続できるモニターが２台たりないので、もってこれる人はもってきて欲しいですー。
僕は１台もっていきます。
@kotatuneco さん、もってこれます？

vestige, 2018-07-22:
作りかけだけど、こういう感じで用意してみてるところです

https://drive.google.com/open?id=143xMlKXmPltiOTmbKHqmZfzh5bRoK18u

akimach, 2018-07-22:
@vestige それか、HDMI オス to VGAメスのケーブルがあればOKです。

kotatuneco, 2018-07-22:
@vestige 
了解しました。
モニタ持参します。

akimach, 2018-07-24:
当日の駐車場所についての連絡です。北側のゲートから入ってください。

午前中にスタッフ用に10組、午後に参加者10組分の駐車することを伝えておきました。

[image]

akimach, 2018-07-24:
補足です。9:30〜14:00まで北ゲートを開けてもらっています。出るのは自由です。

vestige, 2018-07-24:
いろいろ、書き直してみました。ちょっと冗長になったかも...
あと、CM用の説明文ほしいですー。

https://drive.google.com/open?id=143xMlKXmPltiOTmbKHqmZfzh5bRoK18u

mame-n, 2018-07-25:
https://www.instagram.com/p/BllIX2QBSNg/?taken-by=mame_shou
これの問題は、
1. ロボットにつける時、大きな穴を開けて裏から付けるしかできない
2. 手持ちの抵抗が1/6Wとリードが細いので、ブレッドボード・ジャンパーワイヤのメスに挿してもブカブカ
1は裏からつけてもらう、2は1/4Wを米澤さんに買ってきてもらう、でしょうか。
制作上や耐久性の問題はなさそうです。

mame-n, 2018-07-25:
CM用の説明文に那須ラボがあるの、星野さんはご存知かな?

naominix, 2018-07-25:
3Aアダプタお渡しする際にチラシをお渡ししたいと思います

vestige, 2018-07-28:
1/4wの抵抗かいましたー。
あと、 @fleg949 @rohatodenirou さん、資料くださいー。
那須ラボの件はりょうかいです。

fleg949, 2018-07-28:
@vestige 遅くなってごめんなさ～い。Messenngerで送りました。宜しくお願いします<m(__)m>

rohatodenirou, 2018-07-29:
@vestige 遅くなりすみません。情報送りました。よろしくです。

vestige, 2018-07-29:
いろんな人にレビューしてもらって、いちおうこんな感じになりました

https://drive.google.com/open?id=143xMlKXmPltiOTmbKHqmZfzh5bRoK18u

akimach, 2018-07-29:
「HDMI対応」のディスプレイはありますが、HDMIケーブルはないっス。

一度、みなさんのHDMIケーブルの持参の有無を確認しておいたほうがいいと思います。

vestige, 2018-07-30:
@akimach 

何本たりないのかな？
モニターたりない話もあったから確認できてないと思うけど。

fleg949, 2018-07-30:
https://drive.google.com/open?id=143xMlKXmPltiOTmbKHqmZfzh5bRoK18u
宣伝ありがとうございます。

一点、
「CoderDojo」が正しい表記となるので、
スペースなしでお願いします。m(_ _)m
https://coderdojo.jp/docs/brand-guidelines

akimach, 2018-07-30:
@vestige 10本ないです。

vestige, 2018-07-30:
当日のレイアウト

[image]

mame-n, 2018-07-30:
HDMIはA-Aが2本、A-C変換が1つあるよ。必要なら動作確認するけど。

vestige, 2018-07-30:
@mame-n  一応、１０本そろいましたが、念のためもってきてくださいー

mame-n, 2018-07-30:
Foodはお菓子を持っていけるかも。Drinkはコップとか用意しますか?

[15 more replies]

closed 2018-08-16 by akimach