TochigiRaspberryPICircle/kit #37: ２回目の子供向けイベントやります

vestige, 2018-10-21:
https://ppu.connpass.com

vestige, 2018-10-23:
- 事前に用意するものを書き出す
- やることリストを書き出す

vestige, 2018-10-23:
とりあえず、こんなところかな？
参加費をお菓子と飲み物、場所代として一人、500円もらおうと思うんだけど、どうかな？

vestige, 2018-10-29:
お手伝いしてもらえる人はこちら
https://torpc.connpass.com/event/106009/

すでにいっぱいになってた。抽選をしようと思いますー
https://ppu.connpass.com/event/106408/

vestige, 2018-11-06:
いまのところ、８人で止まっているので、抽選しようとおもったけど、このまま８人でうまくやれないかな？
問題はモニターとラズパイ だよね？

ということで、現状の状態を考えてみる

* よねざわ
```
ラズパイ  ３台（3、3B+、 2B+ ）  
モニター  １台
キーボード １台
マウス １台
```

* たかたさん/ふじわらさん

* たかいさん

* なかうちさん

* すがぬまさん

vestige, 2018-11-11:
荷物的な整理も考えても８人は結構、厳しいかなぁ。
モニターがそろわなそう。

kotatuneco, 2018-11-11:
機材：
　ラズパイｘ３
　モニタｘ２
　キーボードx２
　マウスx２

mame-n, 2018-11-11:
ラズパイ x 1
キーボード x 1
マウス x 1

vestige, 2018-11-25:
・日本語のキーボード
・１００均のスピーカー

vestige, 2018-11-25:
PWDでおとなの力をみせる！

vestige, 2018-12-09:
マウスと、キーボード（日本語）を３セットずつ用意しました。

電源タップってみんなが持ってきてくれたら、足りるかな？

vestige, 2018-12-09:
駐車場は２組、連絡がきました

suganuma-kazune, 2018-12-11:
マウス、キーボード　１セット　電源タップも持参します。
１６日よろしくお願いします。

＊＊＊＊＊＊＊＊＊＊＊＊＊＊＊
　　　菅沼　一夫
　　宇都宮市上戸祭1-1-26

　suga320@snow.ucatv.ne.jp
＊＊＊＊＊＊＊＊＊＊＊＊＊＊＊

From: Makoto Yonezawa 
Sent: Sunday, December 09, 2018 9:14 PM
To: TochigiRaspberryPICircle/kit 
Cc: Subscribed 
Subject: Re: [TochigiRaspberryPICircle/kit] ２回目の子供向けイベントやります (#37)

マウスと、キーボード（日本語）を３セットずつ用意しました。

電源タップってみんなが持ってきてくれたら、足りるかな？

vestige, 2018-12-11:
菅沼さん、ありがとうですー。メールの内容がみんな載ってたので編集しておきましたー。

vestige, 2018-12-11:
https://www.raspberrypi.org/documentation/usage/gpio/scratch1/README.md

mame-n, 2018-12-16:
印刷しておこうかな。時間があれば。
https://www.bigmessowires.com/wp-content/uploads/2018/05/Raspberry-GPIO.jpg

fleg949, 2018-12-16:
ごめんなさい。
日本語キーボード2台中1台が、
推定0.1㌧による物理破壊にて
使用できず<m(_　_)m>
キーボードは、日本語x１、英語x１を持参します。
それ以外の機材は、無事です。
電源タップを沢山 二個口を6ぐらい

vestige, 2018-12-16:
キーボードは3台新品があるから大丈夫ですよー
GPIOの印刷あった方がやっぱりいいですかね？

vestige, 2019-01-27:
ひとまずおわったのでクローズ

closed 2019-01-27 by vestige