TochigiRaspberryPICircle/kit #55: 子供向けプログラミングイベント 2023

vestige, 2023-03-26:
7月あたりに今泉小学校でやることになったので必要なものなどあげていく

vestige, 2023-03-26:
・4月の勉強会でmicro:bitつかって、サーボモーターの実験する
・それまでに必要なものを買い揃える
・micro:bitを貸してもらうところを探す
・コミセン側にお金の話を確認する

mame-n, 2023-04-01:
micro:bit、高井さんが20台近く貸し出し可能とのことです。CoderDojo Japanからは10台ということで、私の持っている10台とロットも同じなのは10台となります。

vestige, 2023-04-06:
募集要項っぽいのあとでかきます

はじめてのプログラミング体験
みたいなやつ

vestige, 2023-04-21:
https://studio.beatnix.co.jp/kids-it/hardware/chibi-bit/microbit-servomotor/

vestige, 2023-04-21:
↑を試す予定です

vestige, 2023-04-21:
材料費について聞いてきました

全部で3千円くらいなら出せそうだけど、1人千円は出せないとのこと。
参加費として、集めて持ち帰ってもらうのが良いとのことでした。

あと、販売みたいなのは利益あるなしに関わらず厳しいとのことでした。

vestige, 2023-04-23:
今日(4/23)の勉強会でやること

- イベントの全体的な進捗
- microbit使った実験
- 次回までに購入するもの、やっておくこと

vestige, 2023-04-23:
makecodeのURL

旗振りのセンサー
ーボタン
ー加速度
ー光
ー傾き

発表してもらう形式

時間割をきめてみる

vestige, 2023-04-23:
タイムテーブル

- 10:00
   - 今日はなにするか？
   - 完成品をみせる
- 10:30

kotaproj, 2023-04-23:
基盤作成費は思ったよりかかりそうです
以下、輸送費含む

- 試作基板 - 10枚
  - 3900円(29ドル)
- 本番基板 - 25枚
  - 11400円(85ドル)

vestige, 2023-04-23:
買っておきたいもの
https://akizukidenshi.com/catalog/g/gC-13912/

mame-n, 2023-04-24:
ワークショップの流れ

1. Aで右、Bで左
2. 明るいと右、暗いと左
3. 大きな声を出すたびに、右に左に。
4. 大きな声でワイパー
5. 地磁気で北を見つけよう、見つかったらワイパー
6. リズム良く動かそう(トリガーは大きな声)
7. 角度を変えてリズム良く
8. 三三七拍子をやってみよう

vestige, 2023-04-25:
カードエッジコネクター、秋月でオンラインでは扱ってるけど、店頭では扱ってないみたい…長男に見に行ってもらったらそう言われてしまったらしい…

kotaproj, 2023-04-27:
完成品でいいのでカードエッジコネクタをお借りできませんか？

vestige, 2023-04-27:
了解です。あとでTwitterにメッセージしますね。

kotaproj, 2023-04-28:
- ピン割当

| ピン番号 |  ピンの機能  |
| -------- | ------------ |
| P1       | 超音波センサ |
| P2       | 超音波センサ |
| P8       | servo        |

https://coskxlabsite.stars.ne.jp/html/for_students/microbit/common/microbitpins.html

```
P0は、ブザーで使っているため、使用しないこと

(1)自由に使える端子
 P1
 P2
 P8
 P9 ただしmicro:bit v2でのみ使用可能
 P12
 P16
```

- コネクタとピンの関係

[image]

kotaproj, 2023-04-29:
- 回路図
  - [image]
- 見た目
  - [image]

vestige, 2023-04-30:
おー！なんか、かっこいい！